hku-mars/FAST-Calib #3: 运行错误

XiaoHa0914, 2025-04-22:
你好，郑博，我运行您的数据集很快得到外参数据，但是我运行我的数据出现如下报错：
[ INFO] [1745304940.982427077]: Loading the rosbag /home/adv/FAST-Calib/src/pics/77.bag
[ INFO] [1745304941.366376279]: Loaded 12379968 points from the rosbag.
[ WARN] [1745304941.419599417]: 2 marker(s) found, 4 expected. Skipping frame...
[ INFO] [1745304941.629376932]: Filtered cloud size: 391247
[ INFO] [1745304941.673588974]: Filtered cloud size: 225480
[ INFO] [1745304941.771280835]: Plane cloud size: 120692
[ INFO] [1745304943.540268672]: Extracted 10207 edge points.
[ INFO] [1745304943.556750357]: Number of edge clusters: 6
[fast_calib-2] process has died [pid 4778, exit code -11, cmd /home/adv/FAST-Calib/devel/lib/fast_calib/fast_calib __name:=fast_calib __log:=/home/adv/.ros/log/cd7b2c60-1f46-11f0-afb4-f396782ab41f/fast_calib-2.log].
log file: /home/adv/.ros/log/cd7b2c60-1f46-11f0-afb4-f396782ab41f/fast_calib-2*.log

我的标定板有一些反光，跟这个有关系吗，对光线有什么要求吗？

xuankuzcr, 2025-04-22:
我帮你瞅瞅，数据发我一下

XiaoHa0914, 2025-04-22:
> 我帮你瞅瞅,数据发出一
谢谢郑博的回复，此问题已经解决，是因为我的雷达位置太高，扫描不全导致的。但是遇到新的问题，fast-calib标定结果里只识别到了3个二维码，但是RMSE：1.40mm，请问对外参结果有影响吗？bag和png已经发送至您的邮箱，麻烦您帮忙看下。

XiaoHa0914, 2025-04-22:
> 没收到哇，发我这个邮箱 [zhengcr@connect.hku.hk](mailto:zhengcr@connect.hku.hk)

不好意思郑博，刚刚发错了，现在已经正确的发到您的邮箱了。

xuankuzcr, 2025-04-24:
还是没有收到

xuankuzcr, 2025-04-27:
唔，我就按照你给的参数，绿色点是检测marker得到的3d点，结果是正常的
[image]

XiaoHa0914, 2025-04-28:
> 唔，我就按照你给的参数，绿色点是检测marker得到的3d点，有4个，结果是正常的 [image]

好的，谢谢郑博的回答。

closed 2025-04-29 by xuankuzcr