hku-mars/FAST-Calib #7: circles fitted fail

AshadowZ, 2025-04-23:
I customized a calibration board according to the README specifications.

my lidar: ouster os0 64line
my cam: MV-CS050-10UC + 4mm lens

recording scenario:

[image]

[image]

yaml:

```
  cam_fx: 1213.2994511531224
  cam_fy: 1213.2417349087732
  cam_cx: 1202.551610212725
  cam_cy: 1063.0117849956516
  cam_d0: -0.07334545251214651
  cam_d1: 0.08350477702448131
  cam_d2: 0.0019167201292462936
  cam_d3: -0.0005495698132932394
  ......
  x_min: 1.0
  x_max: 2.0
  y_min: -1.5
  y_max: 1.5
  z_min: -1.0 # avia and mid360: -0.5 ouster: -0.8
  z_max: 1.5
```

It appears no circles were successfully fitted, causing the program to crash away.

[image]

[image]

my data: 
链接: https://pan.baidu.com/s/1FZ8j72UcTtFTIhllr45oCw?pwd=pqqy 提取码: pqqy  --来自百度网盘超级会员v5的分享

🙏🙏🙏🙏🙏🙏🙏🙏

xuankuzcr, 2025-04-24:
Currently, only 128-line mechanical LiDAR is supported. In the future, the method of edge extraction based on LiDAR ring numbers will be integrated, allowing LiDAR models with fewer lines to also be utilized.

AshadowZ, 2025-04-25:
Thank you for your reply, looking forward to the code update.

closed 2025-04-25 by AshadowZ

BetterYimi, 2025-06-23:
> Currently, only 128-line mechanical LiDAR is supported. In the future, the method of edge extraction based on LiDAR ring numbers will be integrated, allowing LiDAR models with fewer lines to also be utilized.

郑博您好，请问你觉得96线的LiDAR可能支持吗？

Xuanrrr, 2025-07-01:
> Currently, only 128-line mechanical LiDAR is supported. In the future, the method of edge extraction based on LiDAR ring numbers will be integrated, allowing LiDAR models with fewer lines to also be utilized.

hello, may i ask when this function can be integrated please?

xuankuzcr, 2025-11-17:
I just updated this repo; it now supports low-line LiDAR (see commit 27df29a).