hku-mars/FAST-Calib #17: 请教使用方法或注意事项

tanble, 2025-05-16:
郑博您好：
       请问一下在使用这个项目标定的时候有什么注意事项吗？看了你在FAST-LIVO2提到了这个标定方法，感觉又快又准，所以想试一下。
       现在采集了一组calib data，但是跑不出结果。我对yaml文件上作了以下修改：
             1. 修改成自己的cam 参数
             2. 按照标定板坐标转出的xyz坐标给定了各个轴的min和max值
             3. bag和img 路径
        以下是run result:
        
[image]
        
        这是标定板的坐标：
        
[image]
[image]
[image]
[image]
[image]
[image]
[image]
      修改后的yaml：

[image]

   希望郑博有空的时候帮忙看一下，在此感谢！

135zhangbin, 2025-05-16:
我也遇到一模一样问题 ，请问 大佬解决了不

tanble, 2025-05-16:
一共三组calib data，参考了#18中大佬分享的经验后，只有一组在微调了各轴的阈值后跑出了效果，但是不是很理想。我这可能是标定板的问题，可能需要重新制作一块儿板子。你可以参考#18分享的经验，我也是按照他的分享的步骤来的。

135zhangbin, 2025-05-16:
好的，感谢

sys111111, 2025-05-17:
请问一下这个标定板的坐标系是怎么转出的？  感谢

tanble, 2025-05-19:
@sys111111 这个方法有很多种，我是先在源码中将从bag读取来的点转成PCD文件，之后就可以用pcl_viewer或者是CloudCompare的点选功能查看标定板的坐标了，最后也可以订阅/filtered_cloud看你框出来的标定板是不是完整的，就可以进行微调了