hku-mars/FAST-Calib #20: 复现效果分享&&问题分享

fff-zx, 2025-05-18:
相机是MV-CA013-A0UC
雷达是livox_horizon或者mid360
软件部分：
1.MVS客户端；
2.mvs_ros_driver;
3.Livox-SDK;
4.livox_ros_driver;
5.fast_livo2;
这里的2和4都是用的大佬开源的LIV_handhold里面的，3是Livox官方的，注意安装2之前需要先安装1(因为会用到它的库),在readme有下载链接;

建议&&心得:
标定外参之前一定要先标定好内参，不然会有严重的影响；
1.内参标定的黑白格板,建议单格0.1m(比A4纸效果好很多)，角点8*6的话那么板子应该是0.8*1m(我是用ros-noetic-camera-calibration包进行标定)；
2.内参标定的时候我是站在3m内拍照(如果符合1,否则可能更近)，标定的图片里，各个角度，方向，距离都要有;
3.我外参标定使用的是标准的1*1.4m的标定板，小了可能对相机能否识别marker和雷达能否识别circle都有影响（会在终端里有对应红字提醒）；
4.horizon的外参标定建议在2-3m左右，推荐2.5 2.6米；
实测距离远了会导致相机很难识别到四个marker，近了可能horizon的点云会有较大畸变(官方手册说0.5m-3m有不同程度畸变);
5.另外一个就是yaml参数里的# Distance filter，需要设置合适的xyz，将点云分割直至只有标定板(如果限制的范围给小了可能会导致一直报错(如下图)，这个时候不妨大胆一点,多一点也没关系的拉)；
ps : 这里我在测试的时候，如果将filter的范围限制的很严格(我主要是对前向X严格限制)，虽然这时候理论上我的板子完全在我给的范围内，但是不知道为什么就是识别不到。如下图:

[image]

[image]

6.如果外参标定出来的旋转矩阵和实际大差不差，但是平移矩阵的Z值却有上米的话（如下图），(引用郑博说的)一定是内参没有标定好导致的，建议重新标内参，然后再填入yaml；

[image]

7.如果不用livox转换盒而是自己接m12航插的话，一定要注意线序啊(网上horizon的英文手册线序是错的!  !  ！ 一定要看中文的手册；或者看avia的接线，因为他俩的接线一模一样，我已经帮大家看过了)；
8.Livox-SDK和Livox-SDK2实测可以同时安装(他们安装的库名不同)；
9.注意mvs_ros_driver的yaml里有一个image_scale参数，(如果你的相机分辨率是1280*1024)，这里设置1就是1280*1024,0.5就是640*512；特别注意! ! ! ! ! !这里用0.5和用1标定出来的内参是不一样的，这会影响外参标定以及fast-livo2建图(fast-livo2里也单独有一个scale参数)，注意；

最后记录一下哈哈
2025.04.20完成mid360复现，以下是效果图

[image]

[image]

[image]

[image]

[image]

2025.05.16完成horizon复现，以下是效果图

[image]

[image]

[image]

[image]

[image]

[image]

[image]
硬件图:

[image]

[image]

[image]

最后这个板子是自己画的，插电脑上就可以，仅用来同步相机与雷达，已经实测过mid360和horizon都可以用，Avia理论和horizon一样应该也可以，如果大家需要可以免费送出，我这还有三块多的；
最后在此感谢郑博的解答与帮助，另外祝大家天天开心！！

135zhangbin, 2025-05-28:
大佬请问您用的标定板定做的什么材质的pvc的还是铝塑板子的，求大佬分享pcb 图可以不

fff-zx, 2025-05-29:
> 大佬请问您用的标定板定做的什么材质的pvc的还是铝塑板子的，求大佬分享pcb 图可以不

外参板是木板（密度板）140r，内参板是KT板28r，pcb你收到了吗，没有的话加我wx吧RoYal3483810699

135zhangbin, 2025-05-30:
好的感谢大佬

gulu996, 2025-07-10:
@fff-zx 大佬，我用的mid360+海康相机，雷达水平放置就能识别出四个点，雷达倾斜后（就是跟你图片上的那个差不多）它就识别不到了，怎么办啊😭😭😭

[image]

4S349, 2025-08-02:
> [@fff-zx](https://github.com/fff-zx) 大佬，我用的mid360+海康相机，雷达水平放置就能识别出四个点，雷达倾斜后（就是跟你图片上的那个差不多）它就识别不到了，怎么办啊😭😭😭
> 
> <img alt="Image" width="2000" height="2000" src="https://private-user-images.githubusercontent.com/148213252/464580583-1667379e-a811-48ef-a124-e5a9afc3fe96.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTQxMzQ4ODgsIm5iZiI6MTc1NDEzNDU4OCwicGF0aCI6Ii8xNDgyMTMyNTIvNDY0NTgwNTgzLTE2NjczNzllLWE4MTEtNDhlZi1hMTI0LWU1YTlhZmMzZmU5Ni5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwODAyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDgwMlQxMTM2MjhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04NGYwZmYwNTVlMGI1ZmQ0ZDJhOTNlZWViZTA1MzYxODFkZGUwYWM3NWU1OTU3YTk3MTY3ZWI2M2I3ZTA1YWRkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.E9aAN6hyVb8v_5gUerjWfuAiV9o2yCAghRtIlCLE1bg">

我参考这个#24 解决了，在代码里面printf拟合误差“error”后发现，是有一个圆的拟合误差在0.21，所以被拒掉了，解决办法也是把error阈值加到0.04，最后得到的RMSE是0.0057，在可接受范围内

szjcqu, 2025-08-06:
> > 大佬请问您用的标定板定做的什么材质的pvc的还是铝塑板子的，求大佬分享pcb 图可以不
> 
> 外参板是木板（密度板）140r，内参板是KT板28r，pcb你收到了吗，没有的话加我wx吧RoYal3483810699

请问板子的尺寸一定要是宽1400mm, 高1000mm么，还是可以设置成任意尺寸。只需修改yaml中的marker_size等参数即可？

myjhvytdrt, 2025-10-31:
你好请问雷达倾斜放置怎么纠正点云呢？
我更改了livox_ros_driver2中的yaml文件：
"extrinsic_parameter" : {
"roll": 0.0,  
"pitch": 25.0, 
"yaw": 0.0,
"x": 0,
"y": 0, 
"z": 0
}
在rviz中看到点云修正了。
fastlivo2中的yaml做了相应的修改：
extrin_calib: 
 extrinsic_T: [-0.011, -0.02329, 0.04412]
 extrinsic_R: [ 0.9063078, 0.0000000, -0.4226183,
               0.0000000, 1.0000000, 0.0000000,
               0.4226183, 0.0000000, 0.9063078 ]
输出的点云还是倾斜的。

gulu996, 2025-10-31:
> 你好请问雷达倾斜放置怎么纠正点云呢？ 我更改了livox_ros_driver2中的yaml文件： "extrinsic_parameter" : { "roll": 0.0, "pitch": 25.0, "yaw": 0.0, "x": 0, "y": 0, "z": 0 } 在rviz中看到点云修正了。 fastlivo2中的yaml做了相应的修改： extrin_calib: [...]

fastlivo2的config里面，mid360.yaml中，
uav:
  imu_rate_odom: false
  gravity_align_en: false
把gravity_align_en改为true

myjhvytdrt, 2025-10-31:
好的谢谢，您硬同步的pcb图可以发我一下吗18263027709@163.com