hku-mars/FAST-Calib #26: 标定结果平移量Z值较大问题

lzk17, 2025-06-12:
您好！郑博。
       感谢您开源如此优秀的工作！我在标定的过程中遇到一些问题，想向您请教一下。
       我标定所用的是15cm厚的PVC标定板、mid360雷达、普通的usb相机模组，设备距离标定板距离约2m。标定结果出现Z轴平移量较大的问题，之前记得在哪看到过说大概率是内参的问题，尝试多次进行内参标定，但是内外参标定结果区别不大。后面在查看重投影效果时发现标定结果是将相机标定到了标定板的另外一侧，Z值平移量也与相机到标定板距离的2倍相符。后面排查是排序后lidar的四个圆心点与相机的四个圆心点顺序正好相反导致。目前的排序算法依靠竖直角进行上排角点确定感觉不太可靠，是不是可以直接根据水平角度排序得到。

xl232lx, 2025-08-06:
> 感谢老哥提供的思路，我fork了一个版本，修改了排序方法，同时改进了一下检查函数的部分，目前测试可以正常工作，如果大家也出现了同样的问题可以试试。https://github.com/engine1wu/FAST-Calib

你好，请问我用你优化的代码跑，Z轴平移量还是很大，请问一下还可能是什么原因？

LIHU577, 2025-08-29:
```
if (v01.cross(v12).z() > 0) {  // 如果是顺时针
	// 反转顺序: 0->0, 1->3, 2->2, 3->1
	std::swap((*v)[1], (*v)[3]);
}
```
请教下，`v01.cross(v12).z() > 0`  不是说明 点 是逆时针排列吗？但注释是 ```如果是顺时针```,
我理解这段是指按顺时针排列结果。
有时间请赐教 @lzk17

zengkailiang, 2025-08-29:
> ```
> if (v01.cross(v12).z() > 0) {  // 如果是顺时针
> 	// 反转顺序: 0->0, 1->3, 2->2, 3->1
> 	std::swap((*v)[1], (*v)[3]);
> }
> ```
> 
> 请教下，`v01.cross(v12).z() > 0` 不是说明 点 是逆时针排列吗？但注释是 `如果是顺时针`, 我理解这段是指按顺时针排列结果。 有时间请赐教 [@lzk17](https://github.com/lzk17)

你是正确的。我看了一下代码，确实>0说明是按逆时针排序，改为 <0 更符合QR码的分布顺序。当然，目前这个代码也不影响标定结果就是。