hku-mars/FAST-Calib #31: wrong sorting algorithm for the center points

AshadowZ, 2025-07-05:
I've recently been using fast-calib for multi-sensor calibration and encountered some strange phenomena. For some perspectives, the calculated translation is several meters, which is clearly incorrect. The root cause is that during the sortPatternCenters step, the order of the center points for the radar and the camera didn't align properly. I did some single-point tests and observed that the output order of this sorting algorithm is unstable.
[image]
[image]
The issue lies in two parts. First, using atan2(sqrt(pt->x * pt->x + pt->y * pt->y), pt->z) to calculate the vertical angle to determine whether a point is above or below. Assuming the camera is facing the calibration board, with the four points being (±1, ±1, 2), the Vertical angle calculated for each point is exactly the same. Could this be a mistake? It seems atan2(pt->y, pt->z) would be correct instead. Second, using atan2(pt->y, pt->x) to calculate the horizontal angle to determine whether a point is on the left or right. It doesn't seem right to directly judge the left or right based on the size of atan2(y, x).
[image]
[image]
Using angle-based sorting is not intuitive, so I wrote a version of the sorting algorithm that directly sorts based on the values of x, y, and z. It seems that there aren't any major issues with this approach.
```
void sortPatternCenters(pcl::PointCloud<pcl::PointXYZ>::Ptr pc, pcl::PointCloud<pcl::PointXYZ>::Ptr v, const std::string& axis_mode = "camera") 
{
    if(pc->size() != 4) 
    {
        std::cerr << BOLDRED << "[sortPatternCenters] Number of " << axis_mode << " center points to be sorted is not 4." << RESET << std::endl;
        return;
    }
    
    v->clear();
    v->reserve(4);
    
    // Transform lidar to camera coordinates if needed
    const bool is_lidar_mode = (axis_mode == "lidar");
    if (is_lidar_mode) {
        for (auto& p : pc->points) 
            p = pcl::PointXYZ(-p.y, -p.z, p.x);
    }
    
    // Find top-most point (minimum y)
    int top_pt = 0;
    for (int i = 1; i < 4; ++i) {
        if (pc->points[i].y < pc->points[top_pt].y) {
            top_pt = i;
        }
    }
    
    // Compute distances from top point
    std::array<double, 4> distances;
    const auto& top = pc->points[top_pt];
    for (int i = 0; i < 4; ++i) {
        const auto& p = pc->points[i];
        distances[i] = sqrt(pow(p.x - top.x, 2) + pow(p.y - top.y, 2));
    }
    
    // Find closest and furthest points
    int min_dist = (top_pt + 1) % 4, max_dist = top_pt;
    for (int i = 0; i < 4; ++i) {
        if (i == top_pt) continue;
        if (distances[i] > distances[max_dist]) max_dist = i;
        if (distances[i] < distances[min_dist]) min_dist = i;
    }
    
    // Determine second top point
    const int top_pt2 = 6 - (top_pt + max_dist + min_dist);
    
    // Determine left/right order
    bool is_left_first = pc->points[top_pt].x > pc->points[top_pt2].x;
    const int lefttop = is_left_first ? top_pt : top_pt2;
    const int righttop = is_left_first ? top_pt2 : top_pt;
    const int leftbottom = is_left_first ? min_dist : max_dist;
    const int rightbottom = is_left_first ? max_dist : min_dist;
    
    // Store sorted points
    v->push_back(pc->points[lefttop]);
    v->push_back(pc->points[righttop]);
    v->push_back(pc->points[rightbottom]);
    v->push_back(pc->points[leftbottom]);
    
    // Transform back to lidar coordinates if needed
    if (is_lidar_mode) {
        for (auto& p : v->points) 
            p = pcl::PointXYZ(p.z, -p.x, -p.y);
    }
}
```

AshadowZ, 2025-07-05:
By the way, the assumption of the sorting algorithm is that the camera and radar are combined in a relatively normal way. If the camera is placed normally but the lidar is flipped 180 degrees, it seems that no matter how you write the existing sorting algorithm, it won't align properly. For such a scenario, it might be necessary to come up with some extra tricks, such as aligning the lidar with gravity after ensuring the camera is facing the board properly.

Xuanrrr, 2025-07-12:
I encounter the same problem as you, and i believe the easiest way to handle this problem in enigneering is to sort the wrong sequence on human.